servalproject/openwrt #15: Reduce boot-time of ME2.0 OpenWRT image

gardners, 2017-01-18:
Look at what kernel modules etc we can remove to reduce boot time.
For example:

1. many of the USB modules can probable be removed. 
2. The kernel image could be built without LZMA to save a few seconds of decompression time.

There is almost certainly other stuff we can remove.

nanometrenat, 2017-11-14:
Boot time is already much faster this week - and only one reboot is required after flashing - not sure what further work is still required?

gardners, 2017-11-14:
Agreed that this can be closed, with boot time in the 2 - 3 minute range for primary functions, and 2.5 - 3.5 minutes for ssh login to become active.  The main solution was to move to using a USB memory stick instead of microSD card, and some rearrangement of the boot process that was made to fix other problems. A fresh issue can be opened if boot time remains a problem.

closed 2017-11-14 by gardners